capacitor-community/camera-preview #169: Official Capacitor 3 support — help wanted

mesqueeb, 2021-10-02:
There are two PRs for Capacitor 3 open:

- https://github.com/capacitor-community/camera-preview/pull/154
- https://github.com/capacitor-community/camera-preview/pull/149

Who has the ability to review, merge and publish on NPM?

I'd gladly pay money if this package could be maintained, but I don't know who I should be talking to. Sadly I personally don't write Swift nor Java...

mesqueeb, 2021-10-24:
@arielhernandezmusa can you give at least github commit rights to @pbowyer
I trust he's more invested in cleaning thing up for Capacitor 3 : )

arielhernandezmusa, 2021-10-24:
Perfect !!!

pbowyer, 2021-11-06:
@mlynch @arielhernandezmusa @SalahAdDin @thomasvidas Who has permission to push to NPM? 🙂

arielhernandezmusa, 2021-11-06:
I have permissions to push to npm

bharring, 2021-11-13:
I'm trying to build and bundle the capacitor-3 branch locally, since it is not in npm yet. But I am hitting the following error. Any suggestions would be greatly appreciated.  Thanks! 

```
~/Documents/camera-preview/ios/Plugin.xcodeproj The file “Pods-Plugin.release.xcconfig” couldn’t be opened because there is no such file. (~/Documents/camera-preview/ios/Pods/Target Support Files/Pods-Plugin/Pods-Plugin.release.xcconfig)

~/Documents/camera-preview/ios/Pods/Target Support Files/Pods-Plugin/Pods-Plugin.debug.xcconfig:1:1: unable to open file (in target "Plugin" in project "Plugin")
```

pbowyer, 2021-11-13:
I'm on my phone so a quick response but do you have CocoaPods installed and
setup?

bharring, 2021-11-13:
Thanks for the quick response! I needed to run `pod install` from the ios
dir., which is a step I don't see anywhere in the capacitor docs. Anyway, I
deleted my original comment because it was very off topic once I realized
the problem. Thanks for all your hard work!

mesqueeb, 2021-12-03:
@pbowyer @arielhernandezmusa will there be a release to npm soon? I'd love to be able to use this with Capacitor 3.x : )

pbowyer, 2021-12-03:
@mesqueeb Does this work for you? https://www.npmjs.com/package/@pbowyer/camera-preview It's my first time publishing to NPM so let me know - and as it's my birthday I haven't had time to test the package myself 🥳 

I'll reply to your other comment after the weekend!

arielhernandezmusa, 2021-12-03:
published version @capacitor-community/camera-preview@2.0.0-beta.2 to npm

pbowyer, 2021-12-16:
Thank you everyone for your help in getting this release finished! We're now at a point where 2.0.0 can be released (see #183 ). The documentation needs improving and it'd be great to update the demo project if anyone is looking for some holiday fun 🎄 

Go on, make my Christmas!

closed 2021-12-22 by mesqueeb

nseb, 2022-01-18:
@pbowyer your version works IOS startRecordVideo function ?