capacitor-community/camera-preview #179: Android app crashes on call startPreview method (NullPointerException)

IgorSamer, 2021-11-10:
I just faced in my Crashlytics dashboard a report of a fatal error in an Android 10 device with the following log:

> Fatal Exception: java.lang.NullPointerException: Attempt to invoke virtual method 'void android.hardware.Camera.startPreview()' on a null object reference
>        at com.ahm.capacitor.camera.preview.CameraActivity$4.onPictureTaken(CameraActivity.java:485)
>        at android.hardware.Camera$EventHandler.handleMessage(Camera.java:1412)
>        at android.os.Handler.dispatchMessage(Handler.java:107)
>        at android.os.Looper.loop(Looper.java:214)
>        at android.os.HandlerThread.run(HandlerThread.java:67)

Looking in the `CameraActivity.java` I found at the end of `onPictureTaken` method:

```
} finally {
  canTakePicture = true;
  mCamera.startPreview(); // here is the line 485 that thrown the error
}
```

How can I fix this? Do I have to add only a verification before like `if (mCamera != null) mCamera.startPreview()` or it needs to be done in another way?

pbowyer, 2021-12-16:
Hi @IgorSamer sorry to hear you've had a fatal error. Did you find out what you needed to do to fix the problem?

jcesarmobile, 2023-06-08:
closing since there was no response

closed 2023-06-08 by jcesarmobile