capacitor-community/camera-preview #186: change aspect ratio

arthrotech, 2021-12-20:
Is there a config option to make the camera appear like the default camera? (aspect ratio 4:3 I believe)

I don't need all the buttons - just wanted the default sizing/aspect ratio

The camera-preview appears to use the full screen

[image]

pbowyer, 2021-12-20:
There isn't, but I like the idea of adding this as an option.

nvahalik, 2023-07-11:
If you are coming from `https://github.com/cordova-plugin-camera-preview/cordova-plugin-camera-preview`, you may notice on iOS that your photos are not in the correct aspect ratio. You may want 16:9 but if you, like me, want the wider 4:3 pictures, there is a simple fix.

Adding this to CameraContoller.swift->prepare()->configurePhotoOutput():

    if captureSession.canSetSessionPreset(.photo) {
        captureSession.sessionPreset = .photo;
    }

...right after the `captureSession` is created will clue the capture session in to the fact that your intent is to take a photo. This will ensure that the proper output is provided.

@jcesarmobile If you have a preferred way of adding this into the plugin, I'd be happy to craft a PR for it. The thing I was consider was adding an `intent` parameter to the `start()` method which takes `"photo" | "video"` as parameters. This would be an iOS-only parameter.

Note that on Android we haven't experienced this issue yet.