capacitor-community/camera-preview #191: ios NSMicrophoneUsageDescription reuested when not needed

moblizeit, 2022-01-01:
**Describe the bug**
on ios with capacitor 3 when i open the cameraPreview my app fails with error
"The app's Info.plist must contain an NSMicrophoneUsageDescription key with a string value explaining to the user how the app uses this data."

I only need to capture pictures and do not require sound. so not sure why it is mandatory? how do i disable for this?

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

pbowyer, 2022-01-31:
Thanks for reporting, this is really annoying I agree! I haven't found why Microphone usage is required; hoping someone in the community can advise and provide a pull request to fix 👍

ryaa, 2022-02-02:
This bug has already been fixed for Android (see https://github.com/capacitor-community/camera-preview/issues/65#issuecomment-783324468)

keeskraak, 2022-04-28:
I also run into this problem on iOS. I only need to take pictures, but somehow access to the microphone is needed.

Is there any update on this issue?

pbowyer, 2022-05-28:
This is fixed in version 3.1.0 released today 🥳

closed 2022-05-28 by pbowyer