capacitor-community/camera-preview #192: Tap and Pinch to Zoom React

THSoftware, 2022-01-04:
Has anyone successfully implemented 'Tap to focus' and 'Pinch to Zoom' using Ionic React (not native) with the Capacitor CameraPreview package?

I've found the below link and have updated my CameraController.swift file to include the new gestures but I'm struggling as to how I add these gestures to the front end react pages? I presume I need to add Tap and Pinch gestures to my react page, and then on fire of the event call CameraPreview.Tap() or CameraPreview.Pinch etc

https://github.com/capacitor-community/camera-preview/blob/f252d76c415586ff07342c5870de3a689966ce3e/ios/Plugin/CameraController.swift

I don't want to use the Capacitor Camera plugin as I have designed a brand specific overlay, so need to use the Camera-preview plugin. Any help would be great!

Some environment variables:
Ionic/React v5.0.7
Capacitor v2

Thanks

pbowyer, 2022-01-05:
I have it working for Android using the latest version which hasn't been released to NPM. A PR with iOS support would be welcome as I haven't had time to look at that.

pbowyer, 2022-03-06:
@THSoftware This is now available in version 2.1.0 and I'll close the issue.

closed 2022-03-06 by pbowyer