capacitor-community/camera-preview #196: IOS captureVideo work with capacitor 3

nseb, 2022-01-18:
Hello , 

in all pull requests, does anyone have an IOS version that works for video recording with capacitor 3?

DiegoRodriguez93, 2022-01-20:
hi @nseb , what version of the plugin are you using? and what version on capacitor?

nseb, 2022-01-20:
@DiegoRodriguez93 
Capacitor 3 , I test camera-preview-2.0.0-beta-2 and the version @pbowyer , same resutlt , app crash when i want record

sultanmyrza, 2022-03-01:
My package.json
```json
"dependencies": {
  // other deps 
  "@capacitor-community/camera-preview": "^2.0.0",
  "@capacitor/android": "3.4.1",
  "@capacitor/app": "1.1.0",
  "@capacitor/core": "3.4.1",
  "@capacitor/ios": "3.4.1",
 // other deps 
}
```
[image]

Should I install any other specific commit to get iOS/Android record video working?

Saqib92, 2022-06-18:
Video Recording in IOS in not supported. I want same is there any way to record video on IOS?

sultanmyrza, 2022-06-20:
@Saqib92 just want to share updates, now I'm using [preview-camera](https://github.com/sultanmyrza/preview-camera) which is used in the [capture-lite app](https://github.com/numbersprotocol/capture-lite/tree/master/src/app/features/home/custom-camera) it works with capacitor 3. Can take photos, and record videos both on iOS, and Android. Here is the old [demo](https://youtu.be/1xsrAEWqEh4). Hope this helps.

Saqib92, 2022-06-20:
> @Saqib92 just want to share updates, now I'm using [preview-camera](https://github.com/sultanmyrza/preview-camera) which is used in the [capture-lite [...]

Plugin Not available to install.

install Command:
> npm install @numbersprotocol/preview-camera

Error: 

```
npm ERR! code E404
npm ERR! 404 Not Found - GET https://registry.npmjs.org/@numbersprotocol%2fpreview-camera - Not found
npm ERR! 404 
npm ERR! 404  '@numbersprotocol/preview-camera@*' is not in this registry.
npm ERR! 404 You should bug the author to publish it (or use the name yourself!)
```

sultanmyrza, 2022-06-20:
@Saqib92 can you try `npm install 'numbersprotocol/preview-camera#release-0.0.2-auto-rotate-fix'`?

sultanmyrza, 2022-06-21:
> > @Saqib92 can you try `npm install 'numbersprotocol/preview-camera#release-0.0.2-auto-rotate-fix'`?
> 
> Build is failing after installing this plugin.

1. android builds or iOS builds?
1. can you please share your build logs?

pbowyer, 2022-06-21:
Can you please take this discussion to the `numbersprotocol/preview-camera` repository, thanks 🙂