capacitor-community/camera-preview #230: Camera Preview is Out of Screen on IOS

Saqib92, 2022-07-05:
**Describe the bug**
`CameraPreview()` is out of bond / Screen when application start for the very first time / coming from lock screen. 
When user unlock application preview is out of screen again. 

**To Reproduce**
Steps to reproduce the behavior:
1. Open Application very first time / Lock & Unlock screen when application is running. 
2. got to any other view and coming back to home page is working fine. 

**Expected behavior**
CameraPreview stay on its place do not change position on lock screen or First time application start. 

**Screenshots**
[image]

**Smartphone (please complete the following information):**
 - Device: iPhone
 - OS: iOS 15.5

**Additional context**
Working fine on Android. This bug is happening on IOS only. When user Install application and open first time this is happening or user Unlock their phone this is happening. Coming back to home screen form other screen this is not happening.

sertal70, 2022-07-07:
Hi @Saqib92 I'm getting crazy with issue #221 , but it seems that your app is not affected by that. I have a couple of questions:

1) What version of Capacitor, Ionic and plugin are you using?
2) Can you reproduce the issue with the [demo](https://github.com/capacitor-community/camera-preview/tree/master/demo) app?

Saqib92, 2022-07-09:
> Hi @Saqib92 I'm getting crazy with issue #221 , but it seems that your app is not affected by that. I have a couple of questions:
> 
> 1. What version of Capacitor, Ionic and plugin are you using?
> A: Capacitor 3.6.0, Ionic: 6.1.9, Plugin: 3.1.0
> 2. Can you reproduce the issue with the [demo](https://github.com/capacitor-community/camera-preview/tree/master/demo) app?
> A: Not checked with demo app. 

But to overcome these issues i removed capacitor plugin and used Cordova camera preview. and everything is working fine now.

sertal70, 2022-07-09:
thank you @Saqib92 , I too tried with Cordova camera preview but unfortunately `toBack` option does not work when using that plugin with Capacitor and I need that option in my app.

pbowyer, 2022-07-20:
Hi @Saqib92, sorry to hear you've got problems with the position of the camera. I've tried to reproduce this and I can't make it happen with Capacitor 3.2.5 / Ionic 5. I don't yet have an Ionic 6 app to test with to see if that's changed how the layout works, but will try to upgrade to Capacitor 3.6 soon and see what happens.

Saqib92, 2022-07-21:
> 

Thank you. If i get a chance i will create Ionic 6 and capacitor 3.6+ project for your help.

Saqib92, 2022-08-22:
> Hi @Saqib92, sorry to hear you've got problems with the position of the camera. I've tried to reproduce this and I can't make it happen with Capacitor 3.2.5 / Ionic 5. I don't yet have an Ionic 6 [...]

Still facing same issue on Capacitor 4, ionic 6,

Reproducible  git repo:

[capacitor-preview-camera-test](https://github.com/Saqib92/capacitor-preview-camera-test)

Saqib92, 2022-08-22:
> thank you @Saqib92 , I too tried with Cordova camera preview but unfortunately `toBack` option does not work when using that plugin with Capacitor and I need that option in my app.

did you found a solution for iOS for `toBack`?

sertal70, 2022-08-22:
@Saqib92 no, Cordova Camera Preview is incompatibile with Capacitor, because the framework handles the WebView in a different way than Cordova.

Saqib92, 2022-08-22:
@sertal70 in my application cordova plugin is working fine. jsut toBack is not working in ios.

sertal70, 2022-08-22:
@Saqib92 you're right, with "is incompatible" I was implicitly referring to the `toBack` feature

Saqib92, 2022-08-22:
> @Saqib92 you're right, with "is incompatible" I was implicitly referring to the `toBack` feature

ohh okay. But is the any workaround for this??