capacitor-community/camera-preview #232: iOS, a picture taken in the landscape orientation is captured as a portrait photo.

orest, 2022-07-10:
**Describe the bug**
I might be doing something wrong, but pictures taken using iPhone X in the landscape orientation are captured as portrait photos. Android works correctly. I can not figure out how to take landscape picture in iOS. i tried to play with rotateWhenOrientationChanged without any luck. 

**To Reproduce**
Steps to reproduce the behavior:
1. Start Preview 
```
 const cameraPreviewOptions: CameraPreviewOptions = {
            position: 'rear',
            parent: 'cameraPreview',
            className: 'cameraPreview',
            rotateWhenOrientationChanged: true,
            toBack: true,
            storeToFile: true,
        };

        await CameraPreview.start(cameraPreviewOptions);
```
3. Rotate the phone  into the landscape
4. Capture the photo
```  
const cameraPreviewPictureOptions: CameraPreviewPictureOptions = {
            quality: 90,
        };

        const result = await CameraPreview.capture(cameraPreviewPictureOptions);
```
6. Captured image will be  a portrait image with the picture rotated 90 degrees. Essentially its impossible to capture landscape image in iOS

**Expected behavior**
Captured image should be landscape 

**Smartphone (please complete the following information):**
 - Device:  iPhone X Max
 - OS: 15.5
 - Browser stock browser, safari

**Additional context**
 - Capacitor 3.4.0
 - @capacitor-community/camera-preview - 3.1.0

danglt1998, 2022-07-19:
Same problem

ryaa, 2022-07-19:
Confirming the problem. This PR should fix this problem https://github.com/capacitor-community/camera-preview/pull/235

Can someone install the fix from here and test?
`npm install ryaa/camera-preview#bufgix/232-ios-picture-taken-in-landscape-orientation-is-captured-as-portrait-photo`

orest, 2022-07-20:
Thank you so much! It looks like it's working. I will give it a better test over the weekend. 
Greatly appreciated!

pbowyer, 2022-07-25:
@orest Were you able to test this? I'd like to merge it into the next release if it's working.

orest, 2022-07-25:
Yes, I was able to perform additional testing on iOS and did a quick regression on Android, and everything is working as expected. Thank you, guys! Awesome job!

orest, 2022-07-25:
@pbowyer I think it's ready to be merged. Thank you!

riderx, 2022-08-03:
@pbowyer if you have time to merge :)

closed 2026-10-24 by orest