capacitor-community/camera-preview #236: Uncaught (in promise) TypeError: Cannot read properties of null (reading 'videoWidth')

ch3njust1n, 2022-07-20:
**Describe the bug**
Sample video code does not work. Cannot read videoWidth.

```javascript
const cameraPreviewOptions: CameraPreviewOptions = {
    position: 'rear',
    parent: 'cameraPreview',
    className: 'cameraPreview',
    height: this.height,
    width: this.width
};

CameraPreview.start(cameraPreviewOptions);

const cameraSampleOptions: CameraSampleOptions = {
    quality: 50
};

while(this.isActive) {
    const image = await CameraPreview.captureSample(cameraSampleOptions)
    console.log(image.value)
}
```

**To Reproduce**
Steps to reproduce the behavior:
1. Run sample video code
2. Open in browser

**Expected behavior**
I expected a frame of the video to be captured.

**Desktop (please complete the following information):**
 - OS: macOS Monterey
 - Brave
 - Version 1.41.96 Chromium 103.0.5060.114

**Smartphone (please complete the following information):**
 - Device: Samsung
 - OS: Android Galaxy S10+

**Additional context**
[image]

Zaphod-Beeblebrox, 2022-12-02:
I get an occasional error just like this with my pwa.  Any answers?

mrcashcash, 2023-08-19:
+1
i get this in Capture and captureSample.

venna-kalyani, 2024-04-22:
same error for me as well, any update ??