capacitor-community/camera-preview #243: Capacitor WebPlugin "CameraPreview" config object was deprecated in v3 and will be removed in v4.

kieranbarlow, 2022-08-02:
**Describe the bug**
Capacitor WebPlugin "CameraPreview" config object was deprecated in v3 and will be removed in v4.

5hee75, 2022-09-03:
I have also seen this warning and came looking for a bit more context.  Is web support for the plugin being removed in v4?

Max-Millerr, 2022-09-07:
"Capacitor WebPlugin "CameraPreview" config object was deprecated in v3 and will be removed in v4."

Does that mean the plugin can no longer be used?

pbowyer, 2022-09-20:
I'm hoping someone who uses this plugin or Capacitor on the web can jump in, as I only use it on Android/iOS

closed 2023-06-08 by jcesarmobile

DrDaxter, 2023-08-14:
Hi, same problem here. I'm using: capacitor/core": "5.2.2", and capacitor-community/camera-preview": "^5.0.0". the warning still appears 😕