capacitor-community/camera-preview #247: Camera not working for web (ionic serve) with Capacitor 4

gregoiregentil, 2022-08-21:
Hello,

Thank you for your hard work. I think that the plugin is not working with Capacitor 4 for the "ionic serve". I have:
    "@capacitor-community/camera-preview": "^4.0.0-0",
    "@capacitor/core": "4.1.0",
as I have install @capacitor-community/camera-preview@next

Please let me know if I have done anything wrong or if it's an issue because of the recent Capacitor 4. My ts code is working in Android and I'm "just" doing:

const { CameraPreview } = Plugins;
CameraPreview.start(cameraPreviewOptions);

The error in the browser says:

ERROR TypeError: Cannot read properties of undefined (reading 'start')

so the cameraPreview doesn't seem to exist.

pbowyer, 2022-09-20:
Hi @gregoiregentil, sorry to hear about these problems. It's going to be something to do with Capacitor 4. I don't use the plugin on the web so I'm hoping someone who does can jump in with a fix. Best of luck!

jcesarmobile, 2023-06-08:
Don't use it like this:
```
const { CameraPreview } = Plugins;
CameraPreview.start(cameraPreviewOptions);
```
Use it like this:
```
import { CameraPreview } from '@capacitor-community/camera-preview';
CameraPreview.start(cameraPreviewOptions);
```

closed 2023-06-08 by jcesarmobile