capacitor-community/camera-preview #248: X,Y position incorrect on iPhone 12 device

sebas2906, 2022-09-29:
**Describe the bug**
When I try to set a fixed X,Y position particularily on iPhone 12 (IOS 16.0.2) is not going correctly. The camera just fixes ok on 0,0. On iPhone 8 (IOS 15.6.1) or other devices like android runs correctly for every position. I don't know if is the same for all notch devices or just iPhone 12 or IOS version.
The X,Y reference is on top left corner of the camera at least on the another devices I have mentioned before, so I'm using it to use an html box of reference

**Screenshots**

[image]
[image]

**Smartphone (please complete the following information):**
 - Device: iPhone 12
 - OS: IOS 16.0.2
 - Version 3.1.1

**Additional context**
Is not the last version of camera preview (3.1.1) but in the commits of the new ones there are no mention of this bug. Is not possible to simulate camera on Xcode so is very hard for me to replicate.

ddx32, 2022-11-03:
I'm experiencing the same issue on iPhone X.

Madserker, 2023-01-25:
I was struggling with this nightmare too, found out that if i **multiply my desired position (x,y) by 1.5 in ios** it centers perfectly the image just like in android. I dont know if this happens in all de iOS I only tried in iPhone7 and iPhone 12 and it worked perfect there. 
Also i found this on internet, I don't know if it related to this, but it says 1.5 so.. i don't know
[image]

s20024, 2023-08-31:
Hi....
I'm Japanese, so my English is't good. sry....

I was able to fix it.
Around line 60 of Plugin.swift
```
self.x = call.getInt("x") != nil ? CGFloat(call.getInt("x")!)/UIScreen.main.scale: 0
self.y = call.getInt("y") != nil ? CGFloat(call.getInt("y")!)/UIScreen.main.scale: 0
```
to
```
self.x = call.getInt("x") != nil ? CGFloat(call.getInt("x")!)/2: 0
self.y = call.getInt("y") != nil ? CGFloat(call.getInt("y")!)/2: 0
```

However, I don't understand why it is "2".
If anyone knows, please let me the reason.