capacitor-community/camera-preview #254: Add camera started event

xulihang, 2022-10-19:
There should be an event with the camera preview size triggered when the camera is started.

This is convenient for adding frame analysing ability using this plugin.

VanIseghemThomas, 2022-10-24:
Would like this also to easily detect a device coming from sleep or another app. Sometimes the camera gets disabled and there is no straight forward way to detect all of these cases as far as I know. Just having a way to check the camera's state would be a big deal.

xulihang, 2022-10-24:
I decided to create my own camera plugin based on Dynamsoft Camera Enhancer, which is convenient to use and works on all three platforms: https://github.com/tony-xlh/capacitor-plugin-camera-preview

VanIseghemThomas, 2022-10-24:
Nice! I'll check it out when IOS and Android support is implemented ;)

xulihang, 2022-10-26:
iOS and Android support has been implemented.

mahnuh, 2023-04-27:
You actually do not need such an event, you could just use `await CameraPreview.start()` and then you can be sure that the camera preview is ready. Works for me quite well this way.

xulihang, 2024-01-02:
This plugin also has a started (onPlayed) event: https://github.com/xulihang/capacitor-plugin-camera

closed 2024-01-02 by xulihang