capacitor-community/camera-preview #256: Multi camera support (wide angle / telephoto)

VanIseghemThomas, 2022-10-24:
**Is your feature request related to a problem? Please describe.**
In a sense, for me this plugin does not seem to detect the wide angle camera. I saw one issue opened on this repo asking the same question: https://github.com/capacitor-community/camera-preview/issues/237

**Describe the solution you'd like**
It would be nice to have the option to switch between the normal / telephoto / wide angle camera on IOS and android.

**Describe alternatives you've considered**
There aren't really any good alternatives. This is by far the best implementation available.

Importeer, 2023-02-10:
Any news on this? I already made changes to flip between the normal camera and Ultra Wide camera on iOS. 

Would be better to change the Flip() method to a ChangeCamera(rear | wide | front) method.

noremacsim, 2024-03-25:
Again Camera is defaulting to the secondary rear camera which is blurry. specify which camera would help