capacitor-community/camera-preview #263: Auto Focus is Not Working

selected-pixel-jameson, 2022-11-19:
**Describe the bug**
No matter what I have to click on the screen to bring the subject in to focus. 

**To Reproduce**
Steps to reproduce the behavior:
1. Point the preview at something close.
2. Point the preview at something far away
3. Camera will not adjust focus.

**Expected behavior**
I expect the camera to adjust the focus as I switch subjects without having to touch the screen. At the very least it should auto focus before I take picture.

**Smartphone (please complete the following information):**
 - Device: iPhone 14 Pro / iPhone 8
 - OS: iOS 16 & iOS 15
 - Version 4.0.0

selected-pixel-jameson, 2022-11-19:
https://user-images.githubusercontent.com/28204537/202831097-ebe2189d-325f-47af-8f19-de13dcb4df48.MOV

moblizeitllc, 2023-04-18:
i have the exact same issue., please advise

nicolidin, 2023-07-10:
But can you focus tapping the screen ? @selected-pixel-jameson @moblizeitllc ??