capacitor-community/camera-preview #267: open camera is slow

scr2em, 2022-12-12:
**Describe the bug**
in my angular ionic app calling `start` takes 1.6s in iPhone 11 and 2.2s in iPhone 13 and 4s in iPhone 14 
```js
 await CameraPreview.start();
 ```

**Expected behavior**
I expect it to load faster in newer versions of iphone not the other way.

**Smartphone (please complete the following information):**
 - Device: [iPhone 11, iPhone 13 , iPhone 14]
 - OS: [e.g. iOS16.1]
 - Capacitor v4 
 - package version: 3.1.2

jkennedy1980, 2023-07-05:
This 4 seconds is hurting me. I'm having to work around it with crafty UX. Does anyone know why it's so slow?

scr2em, 2023-07-06:
@jkennedy1980 I haven't figure it out yet. @riderx can you help with this ?

farestayari, 2024-06-09:
i have the same problem, does anyone has a solution for this ?

jkennedy1980, 2024-06-10:
To be honest, I don't know. I ended up creating a private fork because I needed to do some frame-by-frame ML processing. My private fork is faster. I do see this public fork from around that time with a few small changes: https://github.com/general-galactic/camera-preview-ml

Maybe pull this fork and see if the problem is fixed. I think I intended on pushing back to the main repo and forgot. I'm just not sure if this fork has the right fix.

scr2em, 2024-07-05:
@akships I'm using it with angular, and it's still slow

ygyg70, 2025-01-24:
I ham having the same issue :(