capacitor-community/camera-preview #274: NullPointerException: Attempt to invoke virtual method 'boolean android.widget.FrameLayout.dispatchTouchEvent(android.view.MotionEvent)' on a null object reference

AntonWolter, 2023-01-11:
**Describe the bug**
NullPointerException for Android when "toBack" is set to true.

NullPointerException: Attempt to invoke virtual method 'boolean android.widget.FrameLayout.dispatchTouchEvent(android.view.MotionEvent)' on a null object reference

**To Reproduce**
Steps to reproduce the behavior:
1. Integrate this plugin
2. Set "toBack" to true 
3. Running app
4. When opening a screen that is invoking start of camera preview and user immediately clicks on screen when camera is not finished launching, app crashes

**Expected behavior**

App does not crash when clicking to early on screen..

**Additional context**

In the CameraPreview -> setupBroadcast method we invoke  fragment.frameContainerLayout.dispatchTouchEvent(event);
This will crash when fragment.frameContainerLayout is null. This is not being checked accordingly. Please, check the current method below.

I think, there should be `(null != fragment.frameContainerLayout)` added to the if statement.

```
 private void setupBroadcast() {
        /** When touch event is triggered, relay it to camera view if needed so it can support pinch zoom */

        getBridge().getWebView().setClickable(true);
        getBridge()
            .getWebView()
            .setOnTouchListener(
                new View.OnTouchListener() {
                    @Override
                    public boolean onTouch(View v, MotionEvent event) {
                        if ((null != fragment) && (fragment.toBack == true)) {
                            fragment.frameContainerLayout.dispatchTouchEvent(event);
                        }
                        return false;
                    }
                }
            );
    }
```

DayUx, 2023-05-05:
I have the same issue, it's pretty annoying

nvahalik, 2023-09-18:
Can confirm that this works for us!

tradingproject19, 2024-03-06:
Thanks this condition fixed the issue. But I am unable to reproduce this issue. It happens sometime, I don't know why. Please comment on this thread if anyone finds the root cause.

ricardosohn, 2025-11-26:
It still happens, only on the first start of the camera preview.

ryaa, 2026-02-11:
This PR should fix the issue: https://github.com/capacitor-community/camera-preview/pull/414

It’s based on the PR @ricardosohn created (see https://github.com/capacitor-community/camera-preview/pull/399), with a couple of additional safety improvements (listener lifecycle + extra null guards). I tested it on Android and I can no longer reproduce the crash (including tapping immediately on first start).

If anyone can test the fix and confirm it works for them, I’ll cut a new release (v7.0.5) with this patch.

Thanks!