capacitor-community/camera-preview #275: Preview size is not correctly calculated in Portrait mode

johnbyrne7, 2023-01-20:
When I use camera preview in portrait mode, the size/aspect ratio is not correctly calculated.  The side edges of the preview are cut off, compared to what the regular camera shows.

I see these values in the log, which are obviously not correct, as they are the same values for width and height

   optimal preview size: w: 1088 h: 1088

I believe this is because targetHeight is set to surface height, and then used to both discover available camera aspect ratios, and if no suitable one is not found, targetHeight only is used to find a match.

I made a quick test change to set targetHeight to surface width when in portrait mode, leaving the rest of the logic alone, and that seemed to work.

After this change, I see this in the log:

   optimal preview size: w: 1920 h: 1080

and, the preview sides are now visible.

FernetB, 2023-02-13:
Same here, seems impossible to me to get the same size preview that in the capturaSample method (with Capture is even worse)

I don't know what to do and I think this repo is already dead

@pbowyer is this still maintained?

pbowyer, 2023-02-13:
@FernetB It's not maintained. I have no free time at present to merge PRs, let alone give them the proper testing users deserve. And my contract to develop a company's app that uses this plugin has finished, so I don't get any work time to give to it.

riderx, 2023-02-24:
Can you try `@capgo/camera-preview` ? i believe the problem is not there, otherwise i can fix it

HuyHHV, 2023-04-13:
Hello! It's quite late, so I'm not sure if you still need it. I think the width and height options only accept integer values. If you provide a value with a decimal number, it will default to the screen width or height.

pablo-draxu, 2023-07-11:
@riderx, I just tested, and the same problem occurs in @capgo/camera-preview. It would be awesome if you can fix it there.

algopert, 2023-08-06:
Same issue!