capacitor-community/camera-preview #309: Error when call startRecordVideo on Android.

alabandibar1, 2023-08-04:
**Describe the bug**
APP closed with error when call startRecordVideo on Android.

On Android Studio. In Logcat  we can see this error messages:

Capacitor/Plugin                                V  To native (Capacitor plugin): callbackId: 58129015, pluginId: CameraPreview, methodName: startRecordVideo
Capacitor                                       V  callback: 58129015, pluginId: CameraPreview, methodName: startRecordVideo, methodData: {"position":"front","width":640,"height":480}
CameraActivity                                  D  CameraPreview startRecord camera: front width: 640, height: 480, quality: 70
libc                                            W  Access denied finding property "vendor.camera.aux.packagelist"
CameraActivity                                  W  ************orientationHint ***********= 90
MediaRecorder                                   D  constructor+ 0xb40000773592ca80
MediaRecorder                                   D  constructor- 0xb40000773592ca80
CheckPermission                                 E  media_record-code= 24
CheckPermission                                 E  media_record-code= 22
MediaRecorder                                   D  init+ 0xb40000773592ca80
CheckPermission                                 E  media_record-code= 2
CheckPermission                                 E  media_record-code= 23
MediaRecorder                                   D  init- 0xb40000773592ca80
CheckPermission                                 E  media_record-code= 12
AndroidRuntime                                  D  Shutting down VM
AndroidRuntime                                  E  FATAL EXCEPTION: main
                                                                                                    Process:    , PID: 20892
                                                                                                    java.lang.RuntimeException: setAudioSource failed.
                                                                                                    	at android.media.MediaRecorder.setAudioSource(Native Method)
                                                                                                    	at com.ahm.capacitor.camera.preview.CameraActivity.startRecord(CameraActivity.java:865)
                                                                                                    	at com.ahm.capacitor.camera.preview.CameraPreview$2.run(CameraPreview.java:230)
                                                                                                    	at android.os.Handler.handleCallback(Handler.java:942)
                                                                                                    	at android.os.Handler.dispatchMessage(Handler.java:99)
                                                                                                    	at android.os.Looper.loopOnce(Looper.java:240)
                                                                                                    	at android.os.Looper.loop(Looper.java:351)
                                                                                                    	at android.app.ActivityThread.main(ActivityThread.java:8381)
                                                                                                    	at java.lang.reflect.Method.invoke(Native Method)
                                                                                                    	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:584)
                                                                                                    	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1013)
                                W  Attempt to remove non-JNI local reference, dumping thread
Process                                         I  Sending signal. PID: 20892 SIG: 9

**To Reproduce**

const cameraPreviewOptions: CameraPreviewOptions = {
  position: 'front',
  width: window.screen.width,
  height: window.screen.height,
};

CameraPreview.startRecordVideo(cameraPreviewOptions);

**Smartphone (please complete the following information):**
 - Device: Android 13
 - OS: Android 13

dantheman2865, 2023-09-06:
I am also having this issue on a Pixel 7 running Android 13 and a moto g stylus running Android 11.

I've found that by editing `CameraPreview.java` to add the `RECORD_AUDIO` permission, the app will request permission to record audio and then will not crash. You also need to add `<uses-permission android:name="android.permission.RECORD_AUDIO" />` to your app's AndroidManifest.xml file.

Ideally, we could solve this by implementing the `disableAudio` flag in Android too. I've started to look into it, but I haven't found a way to disable audio recording in the [MediaRecorder](https://developer.android.com/guide/topics/media/platform/mediarecorder) API.

r-k-o-l, 2023-12-21:
I have the same issue and can confirm that adding a "record_audio" permission requirement in Java and in the manifest resolves the issue. Will there be a new version of the plugin soon?