capacitor-community/camera-preview #310: Camera preview doesn't work with native platforms, but web works well

Beercokol, 2023-08-09:
**Describe the bug**
Camera preview doesn't work with native platforms, only white screen with out preview

**IOS logs:** 
```
Loading app at capacitor://localhost...
To Native Cordova ->  File requestAllPaths File731812688 ⚡️  [log] - onscript loading complete
["options": []]
To Native Cordova ->  LottieSplashScreen initialAnimationEnded LottieSplashScreen731812689 ["options": []]
⚡️  WebView loaded
⚡️  To Native ->  App addListener 120489526
⚡️  To Native ->  Preferences get 120489527
⚡️  To Native ->  Preferences get 120489528
⚡️  To Native ->  Preferences get 120489529
⚡️  To Native ->  Preferences get 120489530
⚡️  To Native ->  Preferences get 120489531
⚡️  To Native ->  Preferences get 120489532
⚡️  To Native ->  CameraPreview start 120489533
⚡️  TO JS {"value":null}
⚡️  TO JS {"value":null}
⚡️  TO JS {"value":null}
⚡️  TO JS {"value":null}
⚡️  TO JS {"value":null}
⚡️  TO JS {"value":null}
⚡️  TO JS undefined
⚡️  To Native ->  Preferences get 120489534
⚡️  To Native ->  Preferences get 120489535
⚡️  To Native ->  Preferences get 120489536
⚡️  TO JS {"value":null}
⚡️  TO JS {"value":null}
⚡️  TO JS {"value":null}
⚡️  To Native ->  CameraPreview startRecordVideo 120489537
captureSessionIsMissing
ERROR MESSAGE:  {"message":"Capture Session is Missing","errorMessage":"Capture Session is Missing"}
⚡️  [error] - {"message":"Capture Session is Missing","errorMessage":"Capture Session is Missing"}`
```

[image]

**Android logs**
```
08-08 16:35:59.912 D/Codec2Client(20974): Surface configure completed
08-08 16:35:59.912 D/SurfaceUtils(20974): disconnecting from surface 0xb4000071363b8470, reason disconnectFromSurface
08-08 16:35:59.912 D/CCodecBufferChannel(20974): [c2.exynos.h264.decoder#599] MediaCodec discarded an unknown buffer
08-08 16:35:59.912 D/CCodecBufferChannel(20974): [c2.exynos.h264.decoder#599] MediaCodec discarded an unknown buffer
08-08 16:35:59.912 D/CCodecBufferChannel(20974): [c2.exynos.h264.decoder#599] MediaCodec discarded an unknown buffer
08-08 16:35:59.912 D/CCodecBufferChannel(20974): [c2.exynos.h264.decoder#599] MediaCodec discarded an unknown buffer
08-08 16:35:59.912 D/CCodecBufferChannel(20974): [c2.exynos.h264.decoder#599] MediaCodec discarded an unknown buffer
08-08 16:35:59.912 D/CCodecBufferChannel(20974): [c2.exynos.h264.decoder#599] MediaCodec discarded an unknown buffer
08-08 16:35:59.912 D/CCodecBufferChannel(20974): [c2.exynos.h264.decoder#599] MediaCodec discarded an unknown buffer
08-08 16:35:59.912 D/CCodecBufferChannel(20974): [c2.exynos.h264.decoder#599] MediaCodec discarded an unknown buffer
08-08 16:35:59.912 D/CCodecBufferChannel(20974): [c2.exynos.h264.decoder#599] MediaCodec discarded an unknown buffer
08-08 16:35:59.913 W/cr_MediaCodecBridge(20974): Codec released
08-08 16:35:59.923 I/hw-BpHwBinder(20974): onLastStrongRef automatically unlinking death recipients
08-08 16:35:59.923 D/SurfaceUtils(20974): disconnecting from surface 0xb4000071362b6ae0, reason disconnectFromSurface
08-08 16:35:59.923 E/BufferQueueProducer(20974): [MediaCodec.release](id:51ee00000007,api:0,p:-1,c:20974) detachBuffer: BufferQueue has no connected producer
08-08 16:36:04.704 D/BufferPoolAccessor2.0(20974): bufferpool2 0xb40000709630f598 : 0(0 size) total buffers - 0(0 size) used buffers - 11/17 (recycle/alloc) - 6/13 (fetch/transfer)
08-08 16:36:04.705 D/BufferPoolAccessor2.0(20974): evictor expired: 1, evicted: 1
08-08 16:36:05.705 D/BufferPoolAccessor2.0(20974): bufferpool2 0xb4000070962e46c8 : 0(0 size) total buffers - 0(0 size) used buffers - 7/12 (recycle/alloc) - 5/8 (fetch/transfer)
08-08 16:36:05.705 D/BufferPoolAccessor2.0(20974): bufferpool2 0xb4000070963088e8 : 0(0 size) total buffers - 0(0 size) used buffers - 12/18 (recycle/alloc) - 6/14 (fetch/transfer)
08-08 16:36:05.705 D/BufferPoolAccessor2.0(20974): evictor expired: 2, evicted: 2
08-08 16:36:07.628 W/cr_media(20974): Requires MODIFY_AUDIO_SETTINGS and RECORD_AUDIO. No audio device will be available for recording
08-08 16:36:07.690 D/Preview (20974): screen is rotated 0deg from natural
08-08 16:36:07.690 D/Preview (20974): back camera is oriented -90deg from natural
08-08 16:36:07.690 D/Preview (20974): need to rotate preview 90deg
08-08 16:36:07.691 D/CameraActivity(20974): cameraCurrentlyLocked:0
08-08 16:36:07.711 I/CameraManagerGlobal(20974): Connecting to camera service
08-08 16:36:07.712 D/Preview (20974): optimal preview size: w: 2048 h: 1536
08-08 16:36:07.712 D/Preview (20974): optimal preview size: w: 2048 h: 1536
08-08 16:36:07.718 D/Preview (20974): center vertically
[... 109 lines cut ...]
    "react-swipeable": "^7.0.0",
    "react-virtuoso": "^4.1.0",
    "react-zoom-pan-pinch": "^3.0.7",
    "redux-logger": "^3.0.6",
    "shortid": "^2.2.16",
    "styled-components": "^5.3.6",
    "swiper": "^9.1.0",
    "typescript": "^4.5.2",
    "universal-cookie": "^4.0.4",
    "uuid": "^9.0.0",
    "web-vitals": "^1.1.2",
    "workbox-background-sync": "^5.1.4",
    "workbox-broadcast-update": "^5.1.4",
    "workbox-cacheable-response": "^5.1.4",
    "workbox-core": "^5.1.4",
    "workbox-expiration": "^5.1.4",
    "workbox-google-analytics": "^5.1.4",
    "workbox-navigation-preload": "^5.1.4",
    "workbox-precaching": "^5.1.4",
    "workbox-range-requests": "^5.1.4",
    "workbox-routing": "^5.1.4",
    "workbox-strategies": "^5.1.4",
    "workbox-streams": "^5.1.4"
  },
```

ashchan3811, 2023-10-19:
@Beercokol Are you trying to run camera on iOS simulator or Browerstack? If yes then It will not work since camera will not work there. You need to verify on real device.

For android one, did u apply css for background?

Please look at Readme.md - https://github.com/capacitor-community/camera-preview#startoptions

[image]

viking2917, 2024-12-30:
FWIW, I have the same issue. I am running on a real iOS device, and I have the appropriate permissions set. Will see if I can reproduce in a simpler situation...

ryaa, 2025-06-17:
@Beercokol I’m unable to reproduce this issue. Could you please provide a repository or a sample app (for example, based on the example-app) that demonstrates the problem? That would help us investigate further.