capacitor-community/camera-preview #312: storeToFile: true Does not return a path on Android

dev-at-sevenlab-nl, 2023-08-24:
Using:

```ts
CameraPreview.start({
    position: 'rear',
    disableAudio: true,
    storeToFile: true,
    toBack: true,
})

const { value: sample } = await CameraPreview.captureSample({ quality: 100 });
```

The `value` is a Base64 string ONLY on android, on iOS it returns the file path

Version information
 - Android API 32
 - @capacitor-community/camera-preview 5.0.0
 - @capacitor/ios, android, core 5.2.2

Rahul-Whirldata, 2024-05-17:
experiencing the same issue

sulzbals, 2025-02-25:
Same issue here (@capacitor-community/camera-preview 7.0.0).

Going through the source code for both platforms, it looks like this has really not been implemented for Android, only for iOS.

- [iOS'es `CameraPreview.captureSample`](https://github.com/capacitor-community/camera-preview/blob/be93123d40527ad756bb4ab75c1e86e096af5048/ios/Plugin/Plugin.swift#L186C16-L186C29) (checks `storeToFile` at the end of the method and if true, saves image data to a file and returns its path)
- [Android's `CameraPreview.captureSample`](https://github.com/capacitor-community/camera-preview/blob/be93123d40527ad756bb4ab75c1e86e096af5048/android/src/main/java/com/ahm/capacitor/camera/preview/CameraPreview.java#L99) (calls [`CameraActivity.takeSnapshot`](https://github.com/capacitor-community/camera-preview/blob/be93123d40527ad756bb4ab75c1e86e096af5048/android/src/main/java/com/ahm/capacitor/camera/preview/CameraActivity.java#L744) which does not perform any checks and always returns image data as base64 string)

Rahul-Whirldata, 2025-02-26:
I am checking the platform. if its android then using filestorage, I am storing it into a file in cacheDirectory. If it is ios then directly get uri. This way the implementation because linear for all platforms.