capacitor-community/camera-preview #334: iOS still not getting data from stopRecordVideo()

stickypages, 2024-05-04:
**Describe the bug**
I'm trying to get the video file, works on Android, but not on iOS. I noticed some others saying it wasn't working. Still getting 'empty'. Has there been any changes to this? 

**Smartphone (please complete the following information):**
 - iOS
 
**Additional context**
I have tried to wrap this code in a try{} but nothing get's caught. 
I get the first log, but not the others. 

Hoping for some direction. 

`console.log('[camera] Try getting video file path');

const result = await CameraPreview.stopRecordVideo()
const fileName = result.value.substring(result.value.lastIndexOf("/") + 1);
console.log('[camera] fileName: ', fileName); // << Does not log this
console.log('[camera] Result: ', result); // << Does not log this`

asorteberg, 2024-06-06:
I appear to be having the same issue.

maximilien0405, 2024-06-10:
Same issue for me ! On IOS i get nothing in response :/
@arielhernandezmusa @pbowyer

maximilien0405, 2024-06-18:
@asorteberg @stickypages Hey so since it seems like this plugin is not currently maintained, there is this new plugin called ``capacitor-community/video-recorder`` that works pretty well. It was made to fix all of the issues of this plugin related to video recording. It's currently maintained, but the documentation is not perfect. I suggest reading the installation documentation AND my closed issue here https://github.com/capacitor-community/video-recorder/issues/1 where i encounter a few issues (that are now fixed)

https://github.com/capacitor-community/video-recorder

wijzijnweb, 2024-06-25:
Same here.
Also mentioned here: https://github.com/capacitor-community/camera-preview/issues/307

Someone has fixed this for capacitor 5, but still no fix for this package.
There is still a open pull request

https://github.com/capacitor-community/camera-preview/pull/308

forgr-owner, 2025-01-13:
You can get it in our fork: use https://github.com/Cap-go/camera-preview