capacitor-community/camera-preview #344: `capture()` and `startRecordVideo()` are NOT working in Android 14

Saqib92, 2024-09-09:
Throwing error when trying any method:

`{message: 'Camera is not running'}`

```
async recordVid() {
    try {
      let opt: CameraPreviewOptions = {
        position: 'front',
        width: window.screen.width,
        height: window.screen.height
      }
      await CameraPreview.startRecordVideo(opt)
      
      await new Promise(resolve => setTimeout(resolve, 5000))

      const fileUrl = await CameraPreview.stopRecordVideo()
      console.log(fileUrl)
      
    }
    catch (err) {
      console.log(err)
    }

  }
```

Getting this error:

[image]

Added permissions in manifest:

```
<uses-permission android:name="android.permission.CAMERA" />
<uses-permission android:name="android.permission.RECORD_AUDIO" />
<uses-permission android:name="android.permission.MODIFY_AUDIO_SETTINGS" />

<uses-permission android:name="android.permission.READ_MEDIA_VIDEO" />
<uses-permission android:name="android.permission.READ_MEDIA_IMAGES" />
<uses-permission android:name="android.permission.READ_EXTERNAL_STORAGE" />
<uses-permission android:name="android.permission.WRITE_EXTERNAL_STORAGE" />
```

forgr-owner, 2025-01-13:
You can use https://github.com/Cap-go/camera-preview instead :)

Saqib92, 2025-01-14:
> You can use https://github.com/Cap-go/camera-preview instead :)

Yes, I did use that plugin.