capacitor-community/camera-preview #345: getSupportedFlashModes returns null in Android 14 SDK 34

danielcollione, 2024-09-24:
In our project we found a bug that when starting the camera in its rear mode and using the getSupportedFlashModes method we received an array of all available values: off, auto, on, and torch.
When using the flip() method to access the front camera and using flip() again to return to the rear, using the method again the array comes empty.
And if I try to use the setFlashMode method, the application fails even though the cell phone initially supports Flash.
We performed the test on a physical Android 14 device.

I would like to know if this is really a bug or is there any recommendation for using the following plugin?

Versions:
- Android 14 (SDK 34)
- Capacitor 5
- Ionic/Angular 7
- CameraPreview: 5
- Angular 16
- Model: Samsung A54
- ChromeVersion: 128.0.6613.147

[image]
[image]

ryaa, 2024-12-18:
@danielcollione I confirm the problem. The problem is that  the number might not be limited to just the front and back cameras because modern smartphones often come with more than two cameras. If there are more then 2 cameras (for example, on my Samsung A53 there are 4 cameras total) the flip iterates/switches through all the available cameras, so if flip was invoked twice and there were more then 2 cameras available , the second flip did not return to the initial camera. However, if you flip as many times as cameras available on the device, you will be able to get the flash modes as expected.

This PR should fix the problem https://github.com/capacitor-community/camera-preview/pull/360 as this will flip/switch between the main back/front cameras. Can you please check if the fix works as expected and I will release the new build v6.0.2 with the required fix.

ryaa, 2025-02-03:
The fix for the above issue should be included into the next upcoming release v7.0.0

closed 2025-02-03 by ryaa