capacitor-community/camera-preview #348: How to run demo from scratch?

VSymonenko, 2024-10-01:
What should be the steps to run a demo for the Android platform?:

- Clone project:
```bash
git clone https://github.com/capacitor-community/camera-preview.git
```

- go to the demo directory
```bash
cd ./camera-preview/demo
```

- Install dependencies:
```bash
npm I
```

- run the project on a device or emulator
```bash
npx cap run android
```

Am I right?

jcesarmobile, 2025-03-07:
The example app is now in `example-app` repository.

The steps are right, but you are missing a `npm run build` before `npx cap run android`.
Alternatively you can run `npx cap open android` to open Android Studio and run the app from there.

closed 2025-03-07 by jcesarmobile

ryaa, 2025-03-09:
The option is to run `"build:dev:android:sync:open` or `"build:dev:ios:sync:open` accordingly (see https://github.com/capacitor-community/camera-preview/blob/master/example-app/package.json#L18 and https://github.com/capacitor-community/camera-preview/blob/master/example-app/package.json#L21)