capacitor-community/camera-preview #355: Support for element type: 'video' | 'canvas' and control of that element.

jewel998, 2024-12-15:
**Describe the solution you'd like**
If we could get control over the video or canvas element while the element is being shown, it could help us with several features like rendering AR, VR, Effects, Filters functionality.

**Describe alternatives you've considered**
It would provide us good support for development if MediaStream would be available, in that case, we can use several other features passing it through Web Tensorflow models to benefit better.

Saqib92, 2025-01-14:
https://github.com/VirtuoWorks/CanvasCameraPlugin

jewel998, 2025-01-15:
Thanks, I'll check it out!

jewel998, 2025-01-15:
@Saqib92  thanks, I'll come up with a new capacitor plugin for camera reusing the code and update it to capacitor.

jewel998, 2025-01-16:
@Saqib92 just FYI, the problem with this is the latency caused by high quality, there seems be to be a lot of frame delays or drops, which is not feasible to be able to reciprocate. I've tried it for android [in this plugin](https://www.npmjs.com/package/@jewel998/camera-preview). But the delay is way too much for me to proceed in this direction.

Saqib92, 2025-01-16:
> [@Saqib92](https://github.com/Saqib92) just FYI, the problem with this is the latency caused by high quality, there seems be to be a lot of frame delays or drops, which is not feasible to be able to [...]

Yes, the plugin is old and i used it in my app back in 2022 and it worked. I used it with tensorflowjs to show skeleton on users body. Now i am also looking for updates as i have no experience in Native development. Hope you can manage to make it work. Good luck.

jewel998, 2025-01-17:
I'm also looking for the same kind of use case.
The probable solution I have in my mind is to be able to integrate some AI Model that detects facial or positional landmarks somewhat like Google Sceneform, and has an API to be able to manipulate the GlSurfaceView to render GLTF or obj models.
Previously I had integrated Google Mediapipe but that didn't help a lot when rendering GLTF or OBJ models with ease.

bokzor, 2025-01-25:
* AR Component: Overlays images on camera preview
 * 
 * Features:
 * - Real-time camera preview with image overlay
 * - Camera switching (front/rear)
 * - Photo capture
 * - Cross-platform support (web/native)
 * 
 * Uses Capacitor Camera Preview plugin for native integration
 * and canvas for rendering

`import {
  AfterViewInit,
  ChangeDetectionStrategy,
  Component,
  ElementRef,
  input,
  OnDestroy,
  output,
  viewChild,
  effect,
  signal,
} from '@angular/core';
import { NgIf } from '@angular/common';
import { CameraPreview } from '@capacitor-community/camera-preview';
import { SoundTypeEnum } from '../../models/Sound';
import { PlaysSoundOnClickDirective } from '../../directives/plays-sound-on-click.directive';
import { Capacitor } from '@capacitor/core';

export interface OverlayOptions {
  image: string | undefined;
  top: number;
  height: number;
  left: number;
  width: number;
}

export type FacingMode = 'user' | 'environment';

@Component({
  selector: 'app-ar',
  imports: [PlaysSoundOnClickDirective, NgIf],
  templateUrl: './ar.component.html',
  styleUrl: './ar.component.scss',
  changeDetection: ChangeDetectionStrategy.OnPush,
})
export class ArComponent implements OnDestroy, AfterViewInit {
  readonly overlayOptions = input<OverlayOptions | null>(null);
  readonly cameraOrientation = input<FacingMode>('environment');
  readonly mainCanvas = viewChild<ElementRef<HTMLCanvasElement>>('mainCanvas');
  readonly imageTakenChange = output<string | null>();

  // Convert to signals for UI bindings
  readonly currentOrientation = signal<FacingMode>('environment');
  readonly animationFrameId = signal<number | null>(null);
  readonly imageTaken = signal<string | null>(null);
  readonly showLoader = signal<boolean>(true);
  readonly isPreviewActive = signal<boolean>(false);

  protected readonly SoundTypeEnum = SoundTypeEnum;
  private pngImage = new Image();

  private cameraPreviewOpts = {
    position: this.cameraOrientation() === 'environment' ? 'rear' : 'front',
    parent: 'camera-preview',
    className: 'camera-preview',
    lockAndroidOrientation: true,
    width: window.innerWidth,
    height: window.innerWidth,
    x: 0,
[... 209 lines cut ...]
    }
    if (this.isPreviewActive()) {
      await CameraPreview.stop();
    }
  }

  async preloadAnnotationImage() {
    return new Promise<void>((resolve) => {
      const overlayOptions = this.overlayOptions();
      if (overlayOptions?.image) {
        console.log('Loading annotation image');
        this.pngImage.src = `${overlayOptions.image}`;
        this.pngImage.setAttribute('crossOrigin', 'anonymous');
        this.pngImage.onload = () => resolve();
        this.pngImage.onerror = () => {
          console.error('Error loading annotation image');
          resolve();
        };
      } else {
        resolve();
      }
    });
  }
}
`