capacitor-community/camera-preview #361: Camera.pickImages not works on iOS 18

dodomui, 2025-01-15:
**Describe the bug**
When using Camera.pickImages in iOS18 receive the following error and cause app restart.
Camera.getPhoto is working just fine.

Anyone experienced this?

{"photos":[{"webPath":"capacitor:\/\/localhost\/_capacitor_file_\/private\/var\/mobile\/Containers\/Data\/Application\/10CF6107-3648-4AB7-9C92-64E37E97D692\/tmp\/photo-6.jpg","exif":{},"path":"file:\/\/\/private\/var\/mobile\/Containers\/Data\/Application\
Error Domain=WKErrorDomain Code=5 "JavaScript execution returned a result of an unsupported type" UserInfo={NSLocalizedDescription=JavaScript execution returned a result of an unsupported type}

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

ryaa, 2025-03-19:
@dodomui It seems that this issue is not related to the `capacitor-community/camera-preview`. May be this is related to the capacitor camera plugin?

ryaa, 2025-03-26:
Close as no feedback received from @dodomui

closed 2025-03-26 by ryaa