capacitor-community/camera-preview #362: feat: Capacitor 7 Support

ckoon-infopro, 2025-01-23:
Capacitor 7 has been released.
[https://capacitorjs.com/docs/updating/7-0](https://capacitorjs.com/docs/updating/7-0)

After upgrading from Capacitor 6 to Capacitor 7, I encountered a syntax error while using this plugin.

```
error: cannot find symbol
            call.error("Camera is not running");
                ^
  symbol:   method error(String)
```

> Plugin's success() and error() methods have been removed, use resolve() and reject() instead.

Source: [https://capacitorjs.com/docs/updating/plugins/7-0](https://capacitorjs.com/docs/updating/plugins/7-0)

ryaa, 2025-02-03:
@ckoon-infopro I have implemented the required changes to support Capacitor 7 (see https://github.com/capacitor-community/camera-preview/pull/366). Can you please install `capacitor-community/camera-preview` v7.0.0 from my branch https://github.com/ryaa/camera-preview/tree/chore/update-to-capacitor-7 and check if this works as expected. Thank you

Missile71, 2025-02-12:
It work fine for me

ckoon-infopro, 2025-02-14:
> [@ckoon-infopro](https://github.com/ckoon-infopro) I have implemented the required changes to support Capacitor 7 (see [#366](https://github.com/capacitor-community/camera-preview/pull/366)). Can [...]

Sorry for the late response. I have tested it, and it is working fine.

ryaa, 2025-02-14:
This ticket is duplicated by https://github.com/capacitor-community/camera-preview/issues/367

ryaa, 2025-02-17:
The build v7.0.0 with Capacitor 7.0.0 support has been released - see https://github.com/capacitor-community/camera-preview/releases/tag/v7.0.0

closed 2025-02-17 by ryaa