capacitor-community/camera-preview #367: CameraPreview.java

joazco, 2025-02-13:
lg 73 `call.error("Camera is not running");` => `call.reject("Camera is not running");`

joazco, 2025-02-13:
https://github.com/capacitor-community/camera-preview/blob/master/android/src/main/java/com/ahm/capacitor/camera/preview/CameraPreview.java#L73

ryaa, 2025-02-14:
@joazco this seems to be a duplicate of https://github.com/capacitor-community/camera-preview/issues/362 and covered there - see https://github.com/ryaa/camera-preview/blob/chore/update-to-capacitor-7/android/src/main/java/com/ahm/capacitor/camera/preview/CameraPreview.java#L73

joazco, 2025-02-14:
> [@joazco](https://github.com/joazco) this seems to be a duplicate of [#362](https://github.com/capacitor-community/camera-preview/issues/362) and covered there - see [...]

Oh sorry i didn't see my bad!

closed 2025-02-14 by joazco

ryaa, 2025-02-17:
The build v7.0.0 with Capacitor 7.0.0 support has been released - see https://github.com/capacitor-community/camera-preview/releases/tag/v7.0.0