capacitor-community/camera-preview #387: iOS - app crashes when plugin stop is called

ygyg70, 2025-06-06:
**Describe the bug**
App crashes when plugin stop is called
The crash report + a fix proposed by AI are included with this pull request:
[pull request with crash report](https://github.com/capacitor-community/camera-preview/pull/386)

**To Reproduce**
Steps to reproduce the behavior:
1. Start camera preview
2. Stop camera preview

**Expected behavior**
Camera preview stop should never cause the app to crash

**Smartphone (please complete the following information):**
 - Device: iPhone13,1
 - OS: iPhone OS 18.3.1

ryaa, 2025-06-17:
@ygyg70 I’m unable to reproduce this issue. Could you please provide a repository or a sample app (for example, based on the example-app) that demonstrates the problem? That would help us investigate further.

ryaa, 2025-07-01:
@ygyg70 Thanks for your work on this!
I’ve reviewed your changes and can confirm that they address several crash-prone patterns in the original implementation by:
- Eliminating force unwraps (!)
- Safely handling missing or uninitialized properties
- Removing orientation observers when they’re no longer needed
- Releasing strong references to views during teardown

Since your PR had merge conflicts, I created a follow-up PR with all your changes carried over, along with some dependency updates: https://github.com/capacitor-community/camera-preview/pull/392

I’ve tested it with the `example-app` and one of the production apps I maintain, and everything appears to be working as expected.

Let me know if you have any questions or feedback!

closed 2025-07-01 by ryaa